berkeley-abc/abc #395: Is there a command to perform AND transformations for choice mapping?

phyzhenli, 2025-03-31:
Hi all,

I read Alan's paper and found that there is a different method to do choice mapping:

[image]

So does the current abc has a command to do this? Thank you.

wjrforcyber, 2025-04-03:
AFAIK, no direct implementation, but you could try extract the implementation 
`void Fraig_ManAddChoices( Fraig_Man_t * pMan, int fVerbose, int nLimit )` 
in [src/proof/fraig/fraigChoice.c](https://github.com/berkeley-abc/abc/blob/master/src/proof/fraig/fraigChoice.c) which indicates choices created by associativity and distributivity.

phyzhenli, 2025-04-07:
Thanks for your advice Jingren! I will try it if needed.

closed 2025-04-16 by phyzhenli