berkeley-abc/abc #401: .input_arrival from BLIF seems to be ignored

tkoenig1, 2025-04-08:
With the following BLIF file for a full adder
```
.model fa.pla
.inputs a b c_in
.input_arrival a 0.0 0.0
.input_arrival b 0.0 0.0
.input_arrival c_in 3.0 3.0
.outputs s c_out
.names a b c_in s
111 1
001 1
010 1
100 1
.names a b c_in c_out
111 1
011 1
101 1
110 1
.end
```
and lib2.genlib from SIS, I get after the commands
```
abc 01> read_blif fa.blif
abc 02> strash
abc 03> resyn2
abc 07> map
abc 08> print_delay
Critical path from PI "c_in" to PO "s":
Level   0 : Primary input "c_in".   Arrival time =   0.0. 
Level   1 :         n8/b    (nor2)    Arrival =   0.7.   I/O times: (0.0, 0.0 -> 0.7)
Level   2 :        n11/a    (inv1x)   Arrival =   1.1.   I/O times: (0.7 -> 1.1)
Level   3 :        n12/c    (nand3)   Arrival =   1.5.   I/O times: (0.4, 0.6, 1.1 -> 1.5)
Level   4 :        n13/a    (nand2)   Arrival =   2.1.   I/O times: (1.5, 1.8 -> 2.1)
Level   5 : Primary output "s".   Required time = -1000000000.0.  Path slack = -1000000000.0.
```

Writing out the BLIF file results in
```
# Benchmark "fa.pla" written by ABC on Tue Apr  8 22:33:56 2025
.model fa.pla
.inputs a b c_in
.outputs s c_out

.and_gate_delay 0.64
.default_input_arrival 0 0
.default_output_required -1e+09 -1e+09

.gate nand2  a=b b=c_in O=new_n6
.gate inv1x  a=new_n6 O=new_n7
.gate nor2   a=b b=c_in O=new_n8
.gate oai21  a1=new_n7 a2=new_n8 b=a O=new_n9
.gate inv1x  a=a O=new_n10
.gate inv1x  a=new_n8 O=new_n11
.gate nand3  a=new_n10 b=new_n6 c=new_n11 O=new_n12
.gate nand2  a=new_n12 b=new_n9 O=s
.gate oai21  a1=new_n8 a2=new_n10 b=new_n6 O=c_out
.end
```
So, both in the output and in the written BLIF file, the arrival time
of c_in seems to have been lost.

alanminko, 2025-04-08:
Please try "read_blif -n fa.blif".

tkoenig1, 2025-04-09:
> Please try "read_blif -n fa.blif".

That works:

```
Critical path from PI "c_in" to PO "s":
Level   4 : Primary input "c_in".   Arrival time =   3.0. 
Level   5 :         n6/a    (xnor)   Arrival =   4.1.   I/O times: (3.0, 0.0 -> 4.1)
Level   6 :         n7/a    (xnor)   Arrival =   5.2.   I/O times: (4.1, 0.0 -> 5.2)
Level   7 : Primary output "s".   Required time = 1000000000.0.  Path slack = 1000000000.0.
```
Thanks!