berkeley-abc/abc #409: segfault on reading a library and blif file, possible stack overflow

tkoenig1, 2025-04-21:
With the attached files, I get a segfault in abc:
```
abc 01> read_lib NangateOpenCellLibrary_typical.lib 
Library "NangateOpenCellLibrary" from "NangateOpenCellLibrary_typical.lib" has 90 cells (35 skipped: 21 seq; 6 tri-state; 8 no func; 10 dont_use).  Time =     1.18 sec
Warning: Detected 2 multi-output gates (for example, "FA_X1").
abc 01> read_blif bcdmult_mapped.blif 
Line 6: Skipping line ".and_gate_delay 31.65".
Line 7: Skipping line ".default_input_arrival 0 0".
Line 8: Skipping line ".default_output_required -1e+09 -1e+09".
Warning: The design contains EXDC network.
==138508== Invalid write of size 8
==138508==    at 0xA2EAC0: Mem_FlexEntryFetch (mem.c:409)
==138508==    by 0x607CE3: Abc_SopRegister (abcSop.c:66)
==138508==    by 0x7A8AC1: Io_MvParseLineNamesBlif (ioReadBlifMv.c:2099)
==138508==    by 0x7A8AC1: Io_MvParse (ioReadBlifMv.c:960)
==138508==    by 0x7AB955: Io_ReadBlifMv (ioReadBlifMv.c:183)
==138508==    by 0x7B48BE: Io_ReadNetlist (ioUtil.c:138)
==138508==    by 0x7B4CD3: Io_Read (ioUtil.c:246)
==138508==    by 0x794FFC: IoCommandReadBlif (io.c:554)
==138508==    by 0x78F6D0: CmdCommandDispatch (cmdUtils.c:157)
==138508==    by 0x789065: Cmd_CommandExecute (cmdApi.c:210)
==138508==    by 0x7D0AD7: Abc_RealMain (mainReal.c:371)
==138508==    by 0x7CE024: main (main.c:11)
==138508==  Address 0x38cc5e20 is not stack'd, malloc'd or (recently) free'd
==138508== 
==138508== 
==138508== Process terminating with default action of signal 11 (SIGSEGV)
==138508==  Access not within mapped region at address 0x38CC5E20
==138508==    at 0xA2EAC0: Mem_FlexEntryFetch (mem.c:409)
==138508==    by 0x607CE3: Abc_SopRegister (abcSop.c:66)
==138508==    by 0x7A8AC1: Io_MvParseLineNamesBlif (ioReadBlifMv.c:2099)
==138508==    by 0x7A8AC1: Io_MvParse (ioReadBlifMv.c:960)
==138508==    by 0x7AB955: Io_ReadBlifMv (ioReadBlifMv.c:183)
==138508==    by 0x7B48BE: Io_ReadNetlist (ioUtil.c:138)
==138508==    by 0x7B4CD3: Io_Read (ioUtil.c:246)
==138508==    by 0x794FFC: IoCommandReadBlif (io.c:554)
==138508==    by 0x78F6D0: CmdCommandDispatch (cmdUtils.c:157)
==138508==    by 0x789065: Cmd_CommandExecute (cmdApi.c:210)
==138508==    by 0x7D0AD7: Abc_RealMain (mainReal.c:371)
==138508==    by 0x7CE024: main (main.c:11)
==138508==  If you believe this happened as a result of a stack
==138508==  overflow in your program's main thread (unlikely but
==138508==  possible), you can try to increase the size of the
==138508==  main thread stack using the --main-stacksize= flag.
==138508==  The main thread stack size used in this run was 8388608.
```
Increasing the stack size up to 2^34 has no effect.

[segfault.zip](https://github.com/user-attachments/files/19837977/segfault.zip)

tkoenig1, 2025-04-22:
One additional remark: "read_blif -n" works.

wjrforcyber, 2025-04-22:
Have you updated to the latest version? I haven't seen any crash yet, maybe show your commit hash/OS/etc?

I tried your case on latest commit:
```bash
➜  segfault abc -c "read_lib /Path/to/the/NangateOpenCellLibrary_typical.lib; read_blif /Path/to/the/bcdmult_mapped.blif; ps"
ABC command line: "read_lib /Users/jingrenwang/Downloads/segfault/NangateOpenCellLibrary_typical.lib; read_blif bcdmult_mapped.blif; ps".

Library "NangateOpenCellLibrary" from "/Users/jingrenwang/Downloads/segfault/NangateOpenCellLibrary_typical.lib" has 90 cells (35 skipped: 21 seq; 6 tri-state; 8 no func; 10 dont_use).  Time =     0.04 sec
Warning: Detected 2 multi-output gates (for example, "FA_X1").
Line 6: Skipping line ".and_gate_delay 31.65".
Line 7: Skipping line ".default_input_arrival 0 0".
Line 8: Skipping line ".default_output_required -1e+09 -1e+09".
bcdmult.txt                   : i/o =    8/    8  lat =    0  nd =   155  edge =    407  area =151.79  delay = 6.00  lev = 6
➜  segfault cd ../Github/abc/ && git log -1 | cat
commit eaf974dec1cb0c9100bf7ff14821d62352b62d59
Merge: 5ea124099 b1734ac29
Author: alanminko <37236958+alanminko@users.noreply.github.com>
Date:   Sat Apr 19 00:15:32 2025 +0700

    Merge pull request #406 from chenjunhao0315/master
    
    rewire clean up
```

alanminko, 2025-04-22:
The crash happens before the EXDC network:
> Warning: The design contains EXDC network.
Currently ABC cannot use EXDCs.  It is better to not include it.

wjrforcyber, 2025-04-23:
> Have you updated to the latest version? I haven't seen any crash yet, maybe show your commit hash/OS/etc?
> 
> I tried your case on latest commit:
> 
> ➜  segfault abc -c "read_lib /Path/to/the/NangateOpenCellLibrary_typical.lib; read_blif /Path/to/the/bcdmult_mapped.blif; ps"
> ABC command line: "read_lib /Users/jingrenwang/Downloads/segfault/NangateOpenCellLibrary_typical.lib; read_blif bcdmult_mapped.blif; ps".
> 
> Library "NangateOpenCellLibrary" from "/Users/jingrenwang/Downloads/segfault/NangateOpenCellLibrary_typical.lib" has 90 cells (35 skipped: 21 seq; 6 tri-state; 8 no func; 10 dont_use).  Time =     0.04 sec
> Warning: Detected 2 multi-output gates (for example, "FA_X1").
> Line 6: Skipping line ".and_gate_delay 31.65".
> Line 7: Skipping line ".default_input_arrival 0 0".
> Line 8: Skipping line ".default_output_required -1e+09 -1e+09".
> bcdmult.txt                   : i/o =    8/    8  lat =    0  nd =   155  edge =    407  area =151.79  delay = 6.00  lev = 6
> ➜  segfault cd ../Github/abc/ && git log -1 | cat
> commit eaf974dec1cb0c9100bf7ff14821d62352b62d59
> Merge: 5ea124099 b1734ac29
> Author: alanminko <37236958+alanminko@users.noreply.github.com>
> Date:   Sat Apr 19 00:15:32 2025 +0700
> 
>     Merge pull request #406 from chenjunhao0315/master
>     
>     rewire clean up

Sorry, it indeed crashes, I did the experiment after I removed the external don't care manually.
Anyway, a proper exit should be provide here.

tkoenig1, 2025-04-23:
> Currently ABC cannot use EXDCs. It is better to not include it.

Do I understand you correctly that EXDCs do not work at all, no matter what the file input format? If so, would it be possible to mention this in the online help for commands like exdc_set and exdc_free?