berkeley-abc/abc #414: Genlib generation issue of asap7_clean

chestercc1997, 2025-05-17:
It seems that using the write_genlib command, the asap_clean.lib cannot correctly parse the driven load, as well as the corresponding phase and rise-fall delay information.
``
read asap7_clean.lib;write_genlib asap7.genlib
``

[image]

[asap7_clean.lib.zip](https://github.com/user-attachments/files/20260615/asap7_clean.lib.zip)

wjrforcyber, 2025-05-19:
You could trigger the `<rise-block-delay>` and `<fall-block-delay>` calculation by passing a `Gain` value, which should passed to command `read_lib`, the `read` command couldn't take this parameter which leads to a unit-delay genlib construction here:
https://github.com/berkeley-abc/abc/blob/916f70058ee54a9903153a5a404335e56a41f154/src/map/scl/sclLibUtil.c#L1115-L1116
 and here
https://github.com/berkeley-abc/abc/blob/916f70058ee54a9903153a5a404335e56a41f154/src/map/scl/sclLibUtil.c#L833

wjrforcyber, 2025-05-19:
Hi @alanminko ! There used to be a warning here 12 years ago:
https://github.com/berkeley-abc/abc/blob/916f70058ee54a9903153a5a404335e56a41f154/src/map/scl/sclLibUtil.c#L1127-L1128
Maybe this should be uncommented to at least give users a hint?

closed 2025-05-21 by chestercc1997